one-ware/OneWare #18: No Nodes Created for Generic Vector Length in Entity

swittlich, 2024-10-15:
When using a generic to define the length of an std_logic_vector in a VHDL entity, no nodes are generated in the software. In this case, the MAX_LENGTH generic is set to 138, so I expect 138 nodes to be created. However, because the vector length is defined via the generic, it appears that no nodes are being generated.

```
entity SoM is
    Generic (
        MAX_LENGTH : integer := 138;  
    );
    port (
        clk : in std_logic;
        outV : out std_logic_vector(MAX_LENGTH -1 downto 0)
    );
end entity;
```
Expected Behavior
138 nodes corresponding to the individual bits of the outV vector should be generated.

Actual Behavior
No nodes are generated for the outV vector due to the use of the MAX_LENGTH generic to define its length.

[image]

swittlich, 2024-10-28:
One possibility would be to read in the VHDL / Verilog code with Yosys and then output it as a json. 
Command for VHDL: 
`yosys -m ghdl -p "ghdl --std=08 <file>.vhd -e <top> ; write_json output_yosys.json"`
Command for Verilog: 
`yosys -p "read_verilog <file>.v; write_json output_yosys.json"`

In both cases, this generates a JSON with the following key. 
modules -> ‘<top level entity>’ -> ‘ports’. Under the key there is a map which has the signal name as key and as value a JSON object which executes the used signals under bits and the direction under direction. So whether it is an input or output

For Example this vhdl code translates to this json: 

```
entity SoM is
    Generic (
        MAX_LENGTH : integer := 138;  -- Maximale Länge des Vektors
        COUNTER_LENGTH: INTEGER := 20
    );
    port (
        clk : in std_logic;
        leds : out std_logic_vector(137 downto 0)
    );
end entity;

```

```
[...]
 "ports": {
        "clk": {
          "direction": "input",
          "bits": [ 2 ]
        },                                                                                                                         
        "leds": {
          "direction": "output",
          "bits": [ 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140 ]
        }
      },
[...]
```

HendrikMennen, 2024-11-09:
The Regex should support generic values now, however Constants defined in packages are not working yet

HendrikMennen, 2024-11-09:
> One possibility would be to read in the VHDL / Verilog code with Yosys and then output it as a json. Command for VHDL: `yosys -m ghdl -p "ghdl --std=08 <file>.vhd -e <top> ; write_json output_yosys.json"` Command for Verilog: `yosys -p "read_verilog <file>.v; write_json output_yosys.json"`
> 
> In both cases, this generates a JSON with the following key. modules -> ‘’ -> ‘ports’. Under the key there is a map which has the signal name as key and as value a JSON object which executes the used signals under bits and the direction under direction. So whether it is an input or output
> 
> For Example this vhdl code translates to this json:
> 
> ```
> entity SoM is
>     Generic (
>         MAX_LENGTH : integer := 138;  -- Maximale Länge des Vektors
>         COUNTER_LENGTH: INTEGER := 20
>     );
>     port (
>         clk : in std_logic;
>         leds : out std_logic_vector(137 downto 0)
>     );
> end entity;
> ```
> 
> ```
> [...]
>  "ports": {
>         "clk": {
>           "direction": "input",
>           "bits": [ 2 ]
>         },                                                                                                                         
>         "leds": {
>           "direction": "output",
>           "bits": [ 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140 ]
>         }
>       },
> [...]
> ```

This approach should be applied automatically if the user has yosys/ghdl available