one-ware/OneWare #32: Visual issues in the terminal tool window

Flojo101, 2024-12-06:
If any text in the terminal is highlighted, the width of the highlighted characters changes. Also, some characters are not shown anymore, but reappear when the text is no longer highlighted.

Related: The box character marking the current cursor position is also rendered inside the current working directory, and any sufficiently long command ( >10 characters) has the last chracters rendered to the right of the cursor, while the parts to the left of the cursor are at least partially obscured by the working directory

Screenshot:
[image]

Steps to reproduce:
Just enter a command or highlight some text

HendrikMennen, 2025-02-04:
This seems to only happen with specific fonts. If you use the default Jetbrains one, it should not happen. Until I have more time to fix this completely, I will just hardcode the font (for the terminal) to the builtin jetbrains one.

hendrikmennen, 2026-02-07:
with the current implementation (and default font) this should not be an issue anymore. Feel free to reopen if issues persist