one-ware/OneWare #34: OneWare crashes when zooming in the Waveformviewer on certain zoomfactor

95lux, 2024-12-13:
When inspecting a .vcd waveform, OneWare crashes and gets unresponsive, after zooming in at a certain zoomfactor.
OneWare then starts consuming lots of memory, and doesnt stop. Apparently a lot of Marker objects get created.

### Steps to Reproduce:
1. Load a `.vcd` file in OneWare.
2. Zoom in until OneWare becomes unresponsive ( `x8`, `x2048`, `x131072` or some other factor, dependend on loaded .vcd file).
3. OneWare becomes unresponsive and consumes increasing memory.

[image]

The problem occurs, when `mainStep < 1`.
This loop never escapes and infinite Markers are created:
https://github.com/one-ware/OneWare/blob/db290c8010d3325933bb51c4a37be0907f2aed74/src/OneWare.WaveFormViewer/Controls/WaveFormScale.cs#L69C5-L71C80

i am on Version 0.20.1.0 x64 Linux.
The issue can be recreated on 0.31 Linux/Windows as well.

HendrikMennen, 2025-01-26:
Can you provide a VCD File where that happens so I can reproduce?

95lux, 2025-02-04:
Sure, here you go:
[waveform.zip](https://github.com/user-attachments/files/18654813/waveform.zip)

I have to add, the file does not matter for me. It crashes on each .vcd, on different zoom factors though.
This .vcd was generated with verilator.

HendrikMennen, 2025-02-04:
I see whats going on now, it only happens when the offset is > 0, thats why I could not reproduce it at first. Working on a fix

commit 268fcfe7 (2025-02-04): fix #34
closed 2025-02-04 by HendrikMennen