one-ware/OneWare #45: Issues building on arm64

SpieringsAE, 2025-04-03:
I am trying to build this project on a windows on arm system (I also want to do it for linux arm later but first things first) and I am running into some issues. I think it might be because I have never worked with a dotnet project before and I am just doing something wrong but have no idea what.

I followed the instructions, download/install dotnet, clone, cd to the right directory and `dotnet run` but I get:

```
D:\dev\git\OneWare\src\OneWare.Terminal\Provider\PseudoTerminalConnection.cs(1,7): error CS0246: The type or namespace name 'VtNetCore' could not be found (are you missing a using directive or an assembly reference?)
D:\dev\git\OneWare\src\OneWare.Terminal\ViewModels\TerminalViewModel.cs(11,7): error CS0246: The type or namespace name 'VtNetCore' could not be found (are you missing a using directive or an assembly reference?)
D:\dev\git\OneWare\src\OneWare.Terminal\ViewModels\TerminalViewModel.cs(12,7): error CS0246: The type or namespace name 'VtNetCore' could not be found (are you missing a using directive or an assembly reference?)
D:\dev\git\OneWare\src\OneWare.Terminal\Provider\PseudoTerminalConnection.cs(5,67): error CS0246: The type or namespace name 'IConnection' could not be found (are you missing a using directive or an assembly reference?)
D:\dev\git\OneWare\src\OneWare.Terminal\Provider\PseudoTerminalConnection.cs(11,31): error CS0246: The type or namespace name 'DataReceivedEventArgs' could not be found (are you missing a using directive or an assembly reference?)
D:\dev\git\OneWare\src\OneWare.Terminal\ViewModels\TerminalViewModel.cs(29,12): error CS0246: The type or namespace name 'IConnection' could not be found (are you missing a using directive or an assembly reference?)
D:\dev\git\OneWare\src\OneWare.Terminal\ViewModels\TerminalViewModel.cs(38,12): error CS0246: The type or namespace name 'VirtualTerminalController' could not be found (are you missing a using directive or an assembly reference?)
D:\dev\git\OneWare\src\OneWare.Terminal\ViewModels\TerminalViewModel.cs(27,13): error CS0246: The type or namespace name 'IConnection' could not be found (are you missing a using directive or an assembly reference?)
D:\dev\git\OneWare\src\OneWare.Terminal\ViewModels\TerminalViewModel.cs(35,13): error CS0246: The type or namespace name 'VirtualTerminalController' could not be found (are you missing a using directive or an assembly reference?)

The build failed. Fix the build errors and run again.
```

My guess is that there is some dependency not being loaded, I found the [VtNetCore project by darrenstarr](https://github.com/darrenstarr/VtNetCore) which probably must be loaded with some mechanism I am unaware of and which is not listed in the build steps. How do I correctly load the dependencies for this project?

```
dotnet --version
9.0.202
```

HendrikMennen, 2025-04-04:
Yes, the terminal implementation for windows is quite old and probably needs to be changed to make this work.
Linux ARM should be no problem.

SpieringsAE, 2025-04-04:
I'm just an idiot, I missed the recursive tag on the git clone. dumbdumbdumb. Pulled in the submodules and now it works.

Sorry for the noise, but yeah windows on arm seems to work fine (it starts at least, have not actually done a project yet)

HendrikMennen, 2025-04-04:
Can you check if the terminal works?

SpieringsAE, 2025-04-04:
There was indeed a crash when I tried to open the terminal: 0x8007000b
Which seems to indicate incorrect architecture, so something x86/x64 is still ending up in there somehow.

HendrikMennen, 2025-04-04:
Once I have access to an windows arm machine, I can try to fix this

SpieringsAE, 2025-04-04:
I will attempt to learn some C# magic over the weekend, maybe I can figure something out

SpieringsAE, 2025-04-06:
for info the stack trace:
```
[00:00:08] An attempt was made to load a program with an incorrect format. (0x8007000B)
System.BadImageFormatException: An attempt was made to load a program with an incorrect format. (0x8007000B)
   at winpty.WinPty.winpty_config_new(UInt64 agentFlags, IntPtr& err)
   at OneWare.Terminal.Provider.Win32.Win32PseudoTerminalProvider.Create(Int32 columns, Int32 rows, String initialDirectory, String command, String environment, String arguments) in D:\dev\git\OneWare\src\OneWare.Terminal\Provider\Win32\Win32PseudoTerminalProvider.cs:line 13
   at OneWare.Terminal.ViewModels.TerminalViewModel.CreateConnection() in D:\dev\git\OneWare\src\OneWare.Terminal\ViewModels\TerminalViewModel.cs:line 105
   at Avalonia.Threading.DispatcherOperation.InvokeCore()
   at Avalonia.Threading.DispatcherOperation.Execute()
   at Avalonia.Threading.Dispatcher.ExecuteJob(DispatcherOperation job)
   at Avalonia.Threading.Dispatcher.ExecuteJobsCore(Boolean fromExplicitBackgroundProcessingCallback)
   at Avalonia.Threading.Dispatcher.Signaled()
   at Avalonia.Win32.Win32DispatcherImpl.DispatchWorkItem()
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam)
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32DispatcherImpl.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.DispatcherFrame.Run(IControlledDispatcherImpl impl)
   at Avalonia.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.StartCore(String[] args)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, Action`1 lifetimeBuilder)
   at OneWare.Studio.Desktop.Program.Main(String[] args) in D:\dev\git\OneWare\studio\OneWare.Studio.Desktop\Program.cs:line 49
```

SpieringsAE, 2025-04-06:
I did just try to build it for linux-arm64 and as you expected that works perfectly fine, tested on arch linux arm