one-ware/OneWare #47: Cannot install the vhdplus plugin

SpieringsAE, 2025-04-09:
I'm trying to install the vhdplus plugin, at first it doesn't seem to appear in the extension browser.
So I tried manually adding the path to the extension in the settings which allows me to install it, however it does not function and this error pops up:

```
Plugin /home/user/OneWareStudio/Packages/Plugins/OneWare.Vhdp failed loading: 
 compatibility.txt not found in plugin folder
```
It seems there is some new packaging standard that this plugin does not yet support?

It also seems to be missing [here](https://github.com/one-ware/OneWare.PublicPackages/blob/main/oneware-packages.json) which is why I guess it doesn't show up in the first place.

Are there plans to make this plugin accessible?

hendrikmennen, 2025-04-13:
The VHDPlus plugin is currently not maintained

SpieringsAE, 2025-04-14:
that is unfortunate, thank you for your response

closed 2025-04-14 by SpieringsAE