one-ware/OneWare #73: Random Crash to desktop

timonix, 2025-07-03:
I got a random crash. Maybe it was due to some hover text. Or I may have right-clicked something

Version: 0.21.5.0 OS: Microsoft Windows 10.0.26100 X64
AvaloniaEdit.Rendering.VisualLinesInvalidException: Exception of type 'AvaloniaEdit.Rendering.VisualLinesInvalidException' was thrown.
   at AvaloniaEdit.Rendering.BackgroundGeometryBuilder.GetRectsForSegmentImpl(TextView textView, ISegment segment, Boolean extendToFullWidthAtLineEnd)+MoveNext()
   at AvaloniaEdit.Rendering.BackgroundGeometryBuilder.AddSegment(TextView textView, ISegment segment)
   at AvaloniaEdit.Editing.SelectionLayer.Render(DrawingContext drawingContext)
   at Avalonia.Rendering.Composition.CompositingRenderer.UpdateCore()
   at Avalonia.Rendering.Composition.CompositingRenderer.Update()
   at Avalonia.Rendering.Composition.Compositor.CommitCore()
   at Avalonia.Rendering.Composition.Compositor.Commit()
   at Avalonia.Media.MediaContext.CommitCompositorsWithThrottling()
   at Avalonia.Media.MediaContext.CompositionBatchFinished(Compositor compositor, CompositionBatch batch)
   at Avalonia.Threading.DispatcherOperation.InvokeCore()
   at Avalonia.Threading.DispatcherOperation.Execute()
   at Avalonia.Threading.Dispatcher.ExecuteJob(DispatcherOperation job)
   at Avalonia.Threading.Dispatcher.ExecuteJobsCore(Boolean fromExplicitBackgroundProcessingCallback)
   at Avalonia.Threading.Dispatcher.Signaled()
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam)
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32DispatcherImpl.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.DispatcherFrame.Run(IControlledDispatcherImpl impl)
   at Avalonia.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.StartCore(String[] args)
   at OneWare.Studio.Desktop.Program.Main(String[] args) in C:\actions-runner\_work\OneWare\OneWare\studio\OneWare.Studio.Desktop\Program.cs:line 49

hendrikmennen, 2025-08-12:
seems to be a threading issue with avaloniaedit. Very hard to reproduce, so any fix is difficult to test

hendrikmennen, 2025-12-19:
Closing this for now, since it did not happen in a while. Maybe it got fixed by a recent avaloniaedit update.

If it happens again please reopen